ros-drivers/video_stream_opencv #3: Resource Not Found

rkalghatgi, 2017-03-31:
Hello,

I have downloaded and installed this ROS package. I was able to build the src with no issues, I also executed the make install command and that did not have any issues. However, when I run the package I see the following errors. Any advice or help is much appreciated. Thanks!

osboxes@osboxes:~/catkin_ws/src/video_stream_opencv-master/launch$ roslaunch video_file.launch 
... logging to /home/osboxes/.ros/log/51af9362-1628-11e7-a84f-080027f953fe/roslaunch-osboxes-4463.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/__init__.py", line 307, in main
    p.start()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 746, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 718, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 682, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 587, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 316, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 329, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 142, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 188, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: video_stream_opencv
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/osboxes/rosbuild_ws/package_dir
ROS path [2]=/opt/ros/indigo/share
ROS path [3]=/opt/ros/indigo/stacks

awesomebytes, 2017-03-31:
Looks like you forgot to source your environment.

The error says it's not finding the package video_stream_opencv. Also, did
you modify the launch file with a video file of yours? Otherwise when you
fix the first problem you'll get another error about that :)

If you are on indigo you can just install the package with apt-get. Kinetic
is on its way, but not yet.

rkalghatgi, 2017-03-31:
Thanks for the help. I am using Indigo on Ubuntu 14.04.

Sorry, I’m a bit of a novice with all of this, what would be the correct source command for this?

I did modify the launch file with an AVI file that I have on my desktop. What will be the error that I will receive with this when the first problem is corrected?

My overall end goal is to create a ros bag of the AVI file. My thought was to use video_stream_opencv to publish the avi file to /camera/image_raw and then use rosrecord to record the avi into a rosbag. Do you think that will work?

Thanks!

awesomebytes, 2017-04-03:
Hello. Install the package from apt-get and that error should disappear. (`sudo apt-get install ros-indigo-video-stream-opencv`). Then you can launch the launchfile you made. It should work. And then you should be able to do `rosbag record /camera/image_raw` to make a rosbag of the video as you pointed.

rkalghatgi, 2017-04-03:
Thank you very much! That worked.

closed 2017-04-24 by awesomebytes