ros-drivers/video_stream_opencv #4: Unable to open V4L devices by device node other than /dev/videoN

martonmiklos, 2017-04-22:
Sometimes it would be beneficial to open the video devices with other device node than the /dev/videoN because udev populates the /dev/v4l/by-id entries like:
/dev/v4l/by-id/usb-CN0CJ3PDPA01_Laptop_Integrated_Webcam_FHD-video-index0

Using these device nodes eliminates the necessity of plugging the cameras in specified order to specified ports in a robot which has multiple USB cameras. 

Unfortunately the current if I pass such a device path as a string I got "Could not open the stream." error.

I have checked the OpenCV code, but unfortunately the OpenCV 2.4.8 which is shipped with the Ubuntu 14.04 does not had any feature to specify the V4L2 device manually. 

Lately an open method with an apiReference argument added see:
https://github.com/opencv/opencv/commit/41d8c4d8790ac46bffec1206daccf5d61fefc0ab

Any suggestions would be highly appericiated with this issue.

awesomebytes, 2017-04-24:
You can add an udev rule to make your camera named as /dev/yourpreferred name. I googled quickly and found someone explaining a way here: https://askubuntu.com/questions/715333/assign-webcam-to-a-specific-dev-video

martonmiklos, 2017-04-24:
Using udev makes sense. I will let you know the results!

martonmiklos, 2017-04-24:
FYI I have just built the latest opencv from git, and it handles v4l device node if I pass it as string even if I pass the by-id device node. I will try to link my node against it and I will let you know the results.

awesomebytes, 2018-04-04:
Hey @martonmiklos are you ok with me closing this issue?

martonmiklos, 2018-04-04:
Hello Sam,
Sure.

closed 2018-04-04 by martonmiklos