ros-drivers/video_stream_opencv #5: adding retry for opening video stream

sgillen, 2017-06-13:
Would it be possible to add a retry or wait period for opening streams? It would be useful for me. It seems like it would be straightforward to add a wait loop around line 152 of video_stream.cpp, with the time to wait passed in as a parameter (with the default being zero I suppose). I don't mind making this change and opening a pull request, I did not see any contribution guidelines. 

Thanks for your time!

awesomebytes, 2017-06-14:
Make a PR :) I guess adding a maximum number of retries, and a waiting times should be ok.

Otherwise note that you could always do something like:
```bash
for i in 1 2; do
    roslaunch your_package yourlaunchfile.launch && break || sleep 5;
done
```

In a bash script to launch the node. That would retry 2 times (add more adding 3 4 5...) and sleep 5s in between retries.

k-okada, 2017-06-14:
or you can use retry tag of roslaunch -> http://wiki.ros.org/roslaunch/XML/node

sgillen, 2017-06-15:
thank you both of you! It seems that using shell scripts to add delays between launches is pretty common in the community, so I believe I will go with that option.

closed 2017-06-15 by sgillen