ros-drivers/video_stream_opencv #9: Resource Not Found

jake3991, 2018-01-12:
I've setup this package in my catkin workspace and used the following terminal commands to compile. 
1. source /opt/ros/kinetic/setup.bash
2. catkin_make

I then try to launch the system with the following command from the launch folder of the package
1. roslaunch webcam.launch

I then get the following error
raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: video_stream_opencv
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share

Any idea what's up here?

k-okada, 2018-01-15:
forget to run `source devel/setup.bash`?

jake3991, 2018-01-15:
Yeah that was exactly it, forgive the ROS newbie! Thanks for commenting back!

closed 2018-01-15 by jake3991