ros-drivers/video_stream_opencv #17: speed

sunrentao, 2018-04-16:
When I use this package, the video publish to ros is always speed up so much, I use the video to slam, so it always lost the track, I don't know why, do you have any idea?

awesomebytes, 2018-04-17:
I understand you are publishing from a video file.

Use the latest version of the package from the master branch and change your set_camera_fps parameter in the launchfile to be the fps you want.

The default set_camera_fps is 30. And the previous released version had no control on how fast you read from a video file, that was added last week (will re-release soon).

(If you don't want to lose any frames, if you don't mind, change your fps parameter and it will drop frames to publish at the rate you want).

sunrentao, 2018-04-17:
yes!! this works, thank you a lot

sunrentao, 2018-04-17:
I find another problem, if  I modify the "camera_name", this set will not work

awesomebytes, 2018-04-17:
@sunrentao That was a real mistake, I just pushed the changes to fix it. Let me know if it works for you (I just tried it myself and it worked).

sunrentao, 2018-04-17:
Yes!! Now it works normally. Thank you :)

closed 2018-04-17 by sunrentao