ros-drivers/video_stream_opencv #25: yaml-cpp dependency missing

michael1309, 2018-09-08:
Hi,

I had to add the "yaml-cpp" as a dependency to the CMakeLists.txt.

Old line:
target_link_libraries(video_stream ${catkin_LIBRARIES} ${OpenCV_LIBRARIES})
changed to
target_link_libraries(video_stream ${catkin_LIBRARIES} ${OpenCV_LIBRARIES} yaml-cpp).

Otherwise I got a linker error searching for yaml-function.

awesomebytes, 2018-12-13:
@michael1309  May I ask in which platform did this happen? (Ubuntu or other OS version and ROS version).

Thank you!

michael1309, 2018-12-13:
Hi,

we can close this issue. After my investigation this issue comes due to a mess between ROS1 and ROS2 library installed on the same machine.

Best regards,
Michael

closed 2018-12-13 by michael1309

awesomebytes, 2018-12-13:
@michael1309 Good to know (I was surprised given the package builds in the buildfarm correctly). Thank you!