ros-drivers/video_stream_opencv #29: ROSNode crashes after connecting from   web_video_server ROSNode

Aansh2, 2019-02-06:
My Configuration - 
1. Have started one web_video_server which is used for publish ROS video topic to browser using link 
2. Have started on video_stream_opencv ROSNode using  roslaunch 
3. Able to see ros video topic link in browser  

Issue : when I click on link one request goes to web_video_server node to get Particular topic video/ image  but the  video_stream_opencv node crashes in 1 sec also doe not provide any log error. 

Q. Do we need to have some video player to get video ? 
Q. Does video_stream_opencv able to steam video ? 

Note: I am using "http://wiki.ros.org/video_stream_opencv" example. I have  also tested without web_video_server what is provided in the example visualize=true, but know luck. 

LOG-
SUMMARY
========

PARAMETERS
 * /videoe6/videoe6_stream/buffer_queue_size: 890
 * /videoe6/videoe6_stream/camera_info_url:
 * /videoe6/videoe6_stream/camera_name: videoe6
 * /videoe6/videoe6_stream/flip_horizontal: False
 * /videoe6/videoe6_stream/flip_vertical: False
 * /videoe6/videoe6_stream/fps: 25.0
 * /videoe6/videoe6_stream/frame_id: videoe4
 * /videoe6/videoe6_stream/height: 100
 * /videoe6/videoe6_stream/loop_videofile: True
 * /videoe6/videoe6_stream/set_camera_fps: 35.0
 * /videoe6/videoe6_stream/video_stream_provider: /bbb-360p.mp4
 * /videoe6/videoe6_stream/width: 100
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /videoe6/
    videoe6_stream (video_stream_opencv/video_stream)

ROS_MASTER_URI=http://localhost:11311

process[videoe6/videoe6_stream-1]: started with pid [53162]
[ INFO] [1549452087.564457100]: Resource video_stream_provider: /bbb-360p.mp4
[ INFO] [1549452087.564543000]: Getting video from provider: /bbb-360p.mp4
[ INFO] [1549452087.572626100]: Video stream provider type detected: videofile
[ INFO] [1549452087.576083700]: Camera name: videoe6
[ INFO] [1549452087.579046500]: Setting camera FPS to: 35
[ INFO] [1549452087.579101800]: Camera reports FPS: 24
[ INFO] [1549452087.583555900]: Setting buffer size for capturing frames to: 890
[ INFO] [1549452087.589854200]: Throttling to fps: 25
[ INFO] [1549452087.594140100]: Publishing with frame_id: videoe4
[ INFO] [1549452087.598559800]: Provided camera_info_url: ''
[ INFO] [1549452087.605323400]: Flip horizontal image is: false
[ INFO] [1549452087.608432000]: Flip vertical image is: false
[ INFO] [1549452087.625571900]: Forced image width is: 100
[ INFO] [1549452087.625615800]: Forced image height is: 100
[ INFO] [1549452087.633463000]: using default calibration URL
[ INFO] [1549452087.633532000]: camera calibration URL: file:///root/.ros/camera_info/videoe6.yaml
[ INFO] [1549452087.634625900]: Unable to open camera calibration file [/root/.ros/camera_info/videoe6.yaml]
[ WARN] [1549452087.634663200]: Camera calibration file /root/.ros/camera_info/videoe6.yaml not found.
[ INFO] [1549452087.634718900]: Opened the stream, starting to publish.
[ WARN] [1549452159.435056700]: No calibration file given, publishing a reasonable default camera info.
[ INFO] [1549452159.435117500]: The image width is: 640
[ INFO] [1549452159.435171800]: The image height is: 360
[videoe6/videoe6_stream-1] process has died [pid 53162, exit code -11, cmd /opt/ros/kinetic/lib/video_stream_opencv/video_stream camera:=image __name:=videoe6_stream __log:=/root/.ros/log/8ff4d67e-29f8-11e9-9eb4-0242ac120002/videoe6-videoe6_stream-1.log].
log file: /root/.ros/log/8ff4d67e-29f8-11e9-9eb4-0242ac120002/videoe6-videoe6_stream-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

awesomebytes, 2019-02-06:
Is your video really in the path (taken from the log):

/bbb-360p.mp4

That looks like a very weird path... You may want to specify a full system
path or use the launchfile magic of $(find package_name)/subpath

Is the video correct? As in, you can play it with some other app?

Have you tried to use:

rosrun video_stream_opencv test_video_resource.py /bbb-360p.mp4

As the wiki states? If that doesn't work means opencv can't open the video,
which may mean there is no support for the format of the video (missing
codecs?) or the video is corrupted somehow.

Aansh2, 2019-02-06:
@awesomebytes  Path is correctalso tested with multiple combination like file:////bbb-360p.mp4. 
There is now probe to run the video using test_video_resource.py file. 
All in all the path and video are correct (log prints Video stream provider type detected: videofile ). Some time I can see one frame image in browser/ image_view window also but in back end it crashes.

awesomebytes, 2020-04-26:
Hello, I know this is superlate (sorry about that) but did you end up getting it to work?

Also, in the last post the path is reported to be `_video_stream_provider:=/video-sample.flv` which is different from the previous example and kinda looks wrong again... are you really putting your files at the root `/` of your filesystem?

If no reply in a while we should probably close this issue. Once again, sorry for not giving any further reply in time.

closed 2021-04-10 by awesomebytes