ros-drivers/video_stream_opencv #32: Invalid UE golomb code

ShubhayuDas, 2019-04-05:
I am using the rtsp_stream.launch file. Upon launching I keep getting a recurring error statement - **'Invalid UE golomb code'**, even though the visualization seems to work fine at first. 
I have not found any general solution for this. However are a few cases, documented, where others have faced got this same message while using opencv for rtsp streaming. **I am looking for a general explanation as to what the message signifies in general?** 

Also it may be relevant to mention, that my visualization also seems to be faulty, in case I choose to flip the frames horizontally and/or vertically. Every other frame or so, the visualization shows the un-filped after the first few seconds.  

Cheers

awesomebytes, 2019-04-09:
About the 'Invalid UE golomb code' I'm sorry to say I don't know what is going on, OpenCV is the backend (which has another backend itself) to decode rtsp streams, that must be an error from that backend.

On the visualization issue, I'm pretty sure it may have to do with this PR: https://github.com/ros-drivers/video_stream_opencv/pull/31 which I should evaluate and test soon.

bsack, 2019-05-31:
Was the 'Invalid UE golomb code' error ever resolved? I'm having the same issue and, while everything seems to work fine despite the error, it's a pretty annoying error message, filling up the terminal.

ShubhayuDas, 2019-06-01:
Nope.

awesomebytes, 2019-06-01:
I googled a bit about the 'Invalid UE golomb code' and... there is nothing we can do from the point of  view of `video_stream_opencv`. The error/warning comes from FFMPEG which is the backend of OpenCV, which is our backend. The stream you are using has some anomaly (by FFMPEG implementation interpretation) and that's the output you get when that happens. If the output bothers you... If it's an error output (I'm not sure) you can add to whichever way you are running the node `2> /dev/null` and lose the error output (probably not recommended, but it may be helpful if you are testing something else). Good luck!

closed 2019-06-01 by awesomebytes