ros-drivers/video_stream_opencv #38: Install problem

NightwishMeng, 2019-04-25:
I try to install video_stream_opencv . when I use 'catkin_make' ,it shows errors:
`Base path: /home/deemo/catkin_mpl
Source space: /home/deemo/catkin_mpl/src
Build space: /home/deemo/catkin_mpl/build
Devel space: /home/deemo/catkin_mpl/devel
Install space: /home/deemo/catkin_mpl/install
####
#### Running command: "make cmake_check_build_system" in "/home/deemo/catkin_mpl/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/deemo/catkin_mpl/devel
-- Using CMAKE_PREFIX_PATH: /home/deemo/catkin_mpl/devel;/opt/ros/kinetic
-- This workspace overlays: /home/deemo/catkin_mpl/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/deemo/catkin_mpl/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - video_stream_opencv
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'video_stream_opencv'
-- ==> add_subdirectory(video_stream_opencv)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/deemo/catkin_mpl/build
####
#### Running command: "make -j1 -l1" in "/home/deemo/catkin_mpl/build"
####
Scanning dependencies of target video_stream_node
[ 20%] Building CXX object video_stream_opencv/CMakeFiles/video_stream_node.dir/src/video_stream_node.cpp.o
[ 40%] Linking CXX executable /home/deemo/catkin_mpl/devel/lib/video_stream_opencv/video_stream
[ 40%] Built target video_stream_node
Scanning dependencies of target video_stream_opencv
[ 60%] Building CXX object video_stream_opencv/CMakeFiles/video_stream_opencv.dir/src/video_stream.cpp.o
/home/deemo/catkin_mpl/src/video_stream_opencv/src/video_stream.cpp:52:51: fatal error: video_stream_opencv/VideoStreamConfig.h: No such file or directory
compilation terminated.
video_stream_opencv/CMakeFiles/video_stream_opencv.dir/build.make:62: recipe for target 'video_stream_opencv/CMakeFiles/video_stream_opencv.dir/src/video_stream.cpp.o' failed
make[2]: *** [video_stream_opencv/CMakeFiles/video_stream_opencv.dir/src/video_stream.cpp.o] Error 1
CMakeFiles/Makefile2:1827: recipe for target 'video_stream_opencv/CMakeFiles/video_stream_opencv.dir/all' failed
make[1]: *** [video_stream_opencv/CMakeFiles/video_stream_opencv.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed
`
I have installed successfully , but this time failed. 
I searched the missing file but still not found.
I am wondering some changed

awesomebytes, 2019-04-25:
My first recommendation is to install it using
apt-get install ros-YOURVERSION-video-stream-opencv

Other than that, the error complains about the new dynamic reconfigure
server that I'm not familiar with yet, sorry.

NightwishMeng, 2019-04-25:
> My first recommendation is to install it using apt-get install ros-YOURVERSION-video-stream-opencv Other than that, the error complains about the new dynamic reconfigure server that I'm not familiar with yet, sorry.
> […](#)

yep, I remember last time I use apt-get , and it works. 
Thx for ur kind very much !

furushchev, 2019-04-25:
@NightwishMeng Sorry for trouble on my latest update for supporting dynamic reconfigure.
It looks that I forgot to add dependency of generated configuration header file to the library which uses the header.
I made a pull request for this issue.
Could you test if this patch fixes the error?

NightwishMeng, 2019-04-25:
@furushchev My pleasure hhhh. I have tried and it workds ! 
`[100%] Built target video_stream_opencv
`

furushchev, 2019-04-25:
@NightwishMeng Thank you for testing!

closed 2019-04-25 by furushchev