ros-drivers/video_stream_opencv #40: Having a issue for video streaming

dohkim119, 2019-04-30:
I'm the one who are using ros package named 'video_stream_opencv'.

While I'm using it, I met a problem getting video stream from the device.

Here is my H/W Spec and situation.

Board : Jetson TX2

Device : Fushicai USBTV007 Video Grabber [EasyCAP]

rosdistro : ROS- Kinetic

OS : Ubuntu 16.04

OpenCV : 3.4.1 (replace 3.3.1-dev cv2.so with 3.4.1 cv2.so)

So I tried to get a video stream through the Easycap.
The './test_video_resource.py' works well at the moment at it shows video stream.
But when I use 'webcam_launch', if keep showing 'VIDIOC_QUERYCTRL: Inappropriate ioctl for device' error message.

So would anyone can give me some advice to solve this problem?

furushchev, 2019-05-07:
Hi,

1. Just clarifying, you cannot see any image topic published? or just the error messages shows up while the image can be retrieved?
1. Could you paste all the output logs when you run `test_video_resource.py` to help debugging?
2. Which version of `video_stream_opencv` do you use? Could you run `rosversion video_stream_opencv` and paste the result if you installed this package via `apt-get`, otherwise `roscd video_stream_opencv && git rev-parse HEAD`?

awesomebytes, 2020-04-26:
No reply in almost a year, I'll close this issue, feel free to reopen it if there is any further info @furushchev @dohkim119 :)

closed 2020-04-26 by awesomebytes