ros-drivers/video_stream_opencv #41: Could not capture frame video_file.launch

HareshMiriyala, 2019-05-08:
I'm able to play a video.avi file using roslanch video_stream_opencv video_file.launch  

However, it doesn't loop the video thought the argument loop_videofile is set to true.

HareshMiriyala, 2019-05-08:
Just checked that when i convert the .avi file to .mp4, it works !

furushchev, 2019-05-09:
@HareshMiriyala Hi, thanks for the report!
For debugging, could you paste all the logs displayed in the terminal you run the node using .avi.

HareshMiriyala, 2019-08-21:
Hi, i donot have access to the project right now. I'm closing this issue for now. For someone looking here for solutions, I was able to fix this problem by having a .mp4 version of the video. Maybe it was a codec issue that it didn't work with the .avi format, but i'm not sure.

closed 2019-08-21 by HareshMiriyala