ros-drivers/video_stream_opencv #46: Selecting pixel format

machinekoder, 2019-06-18:
Is there a way to select the v4l2 cameras pixel format when starting the opencv stream?

It looks like it selects the uncompressed stream per default on my device, which rather slow at 1080p even with USB 3.0.

J-Rojas, 2019-07-18:
The output format should also be selectable. BGR8 is hard-coded.

J-Rojas, 2019-07-22:
@machinekoder OpenCV only guarantees support for reading BGR8 from video streams. See these references: 

https://devtalk.nvidia.com/default/topic/1020915/jetson-tx2/opencv-and-webcam-problem-pixel-format-of-incoming-image-is-unsupported-by-opencv/ 

https://stackoverflow.com/questions/42790290/why-calling-cv2-videocapture-changes-the-pixel-format-of-a-v4l-camera

furushchev, 2019-07-23:
@J-Rojas I think what @machinekoder mentioned about is not about format of published image messages but image data format captured from v4l2 camera device.

@machinekoder Currently since we don't specify any option to set pixel format for v4l2 camera device, OpenCV tries to set possible format in certain order.
( https://github.com/opencv/opencv/blob/f663e8f903645a3dd66f6833f63717b86e861d77/modules/videoio/src/cap_v4l.cpp#L485 )
Looking at the code, it looks we can change the format by setting `cv::CAP_PROP_FOURCC`
( https://github.com/opencv/opencv/blob/f663e8f903645a3dd66f6833f63717b86e861d77/modules/videoio/src/cap_v4l.cpp#L1832 )

marcelino-pensa, 2022-04-27:
I was able to get images from the MJPG stream by adding the following in [this line](https://github.com/ros-drivers/video_stream_opencv/blob/65949bdc5c9468d18c51aed9073d020bec892532/src/video_stream.cpp#L295):
`cap->set(cv::CAP_PROP_FOURCC, cv::VideoWriter::fourcc('M','J','P','G'));`

It worked, but the CPU usage went through the roof... not sure if that's expected, but I'd appreciate if anyone had any ideas on how to overcome high CPU usage, or whether that'll be a given for getting data from a MJPG stream