ros-drivers/video_stream_opencv #48: Failed to open nodelet

ml-boringtao, 2019-08-10:
Successfully show video by running: ./test_video_resource.py videofile
However, when roslaunch video_stream_opencv video_file.launch video_stream_provider:=videofile Error msg shows: 

**[ERROR] [1565440225.383388393]: Failed to load nodelet [/videofile/videofile_stream] of type [video_stream_opencv/VideoStream] even after refreshing the cache:
Failed to load library devel/lib//libvideo_stream_opencv.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libyaml-cpp.so.0.5: cannot open shared object file: No such file or directory)
 [ERROR] [1565440225.383557928]: The error before refreshing the cache was: Failed to load library devel/lib//libvideo_stream_opencv.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libyaml-cpp.so.0.5: cannot open shared object file: No such file or directory)
**

Can you help? thanks

furushchev, 2019-08-19:
Hi,
It seems the nodelet cannot be found in your environment.
I guess you forgot to install necessary packages to be installed (did you run `rosdep install` before `catkin build`?) or you are using non-standard library for dependencies of this library (Are you using yaml-cpp installed by `apt-get`? If not, please use the standard one installable by using `apt-get`).
If everything is ok, the nodelet `video_stream_opencv/VideoStream` should be listed when you run `rosrun nodelet list_nodelets`.
Hope it will help you.

pdisho, 2019-08-22:
Actually the issue is in the CMakeLists.txt when it's attempting to install the video_stream_node. 

"install(TARGETS video_stream_node
    RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}"

This is saying install with the name video_stream_node but there isn't any target to actually install.

"install(TARGETS ${PROJECT_NAME} 
    video_stream_node
    RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
    ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
    LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}"

This fixes it for me because now the target is video_stream_node.

furushchev, 2019-08-23:
@pdisho Nice catch! I made a PR to fix the issue. #51 Please give me a feedback if exists :+1:

furushchev, 2019-08-23:
@ml-boringtao Could you check if the error is fixed by using the PR I opened? #51

GrantDare, 2019-08-28:
Hi @furushchev I can confirm #51 does solve the issue. I also experienced this issue, implemented the changes and now it works.

furushchev, 2019-09-02:
@GrantDare Thank you for confirmation! I've now merged the PR.
@ml-boringtao As for now, I close this issue since the PR #51 was merged.
Please feel free to re-open the issue if it is not yet fixed.

closed 2019-09-02 by furushchev