ros-drivers/video_stream_opencv #56: how to improve the latency sourcing video with rtsp?

D7823, 2019-12-28:
Hi all,

I try to use this driver to stream video through rtsp for a SLAM task. But when I run the testing python script to test the video source, I find the video has a big latency, like about 10 seconds which is intolerable to my project. Is there any method to narrow the latency?

Thanks in advance.
Duncan

furushchev, 2019-12-29:
Could you please check where the latency comes from?
If the latency comes from the node from this repository, then you can check by writing plain opencv code and estimate the time to decode the received video. 
I guess the latency comes from encoding and decoding video stream.

D7823, 2019-12-29:
I think it might do with the python test script so I directly run a launch file modified to my rtsp source, but the latency has no improvement.
Then, I switch to another tool called MPlayer to source the rtsp livestream and this time, the latency is greatly improved to <0.5s. So it might do with this repository or the image-view package? 

Also, there pops out an error message "(image_raw: 15196) g_object_unref: assertion 'G_IS_OBJECT (object)' failed". 
Would it be related?

machinekoder, 2020-01-28:
I'm also experiencing significant lag with this node. I'm not sure how I can determine if it comes from the camera or the OpenCV decoding. Any ideas on how to troubleshoot this problem?

awesomebytes, 2020-01-28:
You can try to build a minimal example of an OpenCV app to open the stream.
Even thought... That's mainly what this repo does with extra options.

If it happens also with the python testing script... It may be that the
backend OpenCV is using is not good for you.

I'd advise to try gscam also.

awesomebytes, 2020-04-26:
Hey @machinekoder or @D7823 , did you find a way to get less latency on the rtsp stream? Either with this node or a different one?

Someone else may be having a similar issue in https://github.com/ros-drivers/video_stream_opencv/issues/64 so it would be nice to see how to deal with this situation (and maybe improve it in this project if there is a way).

D7823, 2020-04-26:
My case may be a bit different. Still using this node, I directly connected two machines through the ethernet cable and setting up the static IP addresses. With this setting, I didn't measure the actual latency but at least no obvious latency observed by bare eyes. 
Hope it helps.

awesomebytes, 2020-04-28:
@D7823 Thanks for the update. It makes a lot of sense that just reducing network delay will make things work more smoothly.

closed 2021-04-10 by awesomebytes