ros-drivers/video_stream_opencv #58: Select Timeout Issue When Testing

nickkraus, 2020-01-18:
After installing, and running ./test_video_resource.py 0
I get the errors below, any ideas? I am using a SainSmart IMX219 Camera Module on Jetson Nano.

Trying to open resource: /dev/video0
Correctly opened resource, starting to show feed.
select timeout
select timeout
OpenCV Error: Assertion failed (total() == 0 || data != NULL) in Mat, file /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/core/include/opencv2/core/mat.inl.hpp, line 431
Traceback (most recent call last):
  File "./test_video_resource.py", line 43, in <module>
    rval, frame = cap.read()
cv2.error: /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/core/include/opencv2/core/mat.inl.hpp:431: error: (-215) total() == 0 || data != NULL in function Mat

awesomebytes, 2020-01-18:
That's an internal error of OpenCV, I'm guessing it's not able to extract
an image (size 0 it reports?) from the device.

Does the camera work with some other software like 'cheese'?

Good luck!

nickkraus, 2020-01-18:
Thanks for the quick reply! Image streams fine if I run:

gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),width=3820, height=2464, framerate=21/1, format=NV12' ! nvvidconv flip-method=0 ! 'video/x-raw,width=960, height=616' ! nvvidconv ! nvegltransform ! nveglglessink -e

from: https://github.com/JetsonHacksNano/CSI-Camera

In addition to this issue, when I run $ roslaunch video_stream_opencv camera.launch, I get a few issues:

1. GLib-GObject-CRITICAL **: 21:49:43.015: g_object_unref: assertion 'G_IS_OBJECT (object)' failed (and G_IS_ELEMENT) and

2. A few moments later I get a bunch of VIDIOC_QBUF: Invalid argument messages

Its probably a system setup issue as opposed to an issue with the git repo. Just thought i'd post to see if you have seen it before.

awesomebytes, 2020-01-18:
There is a different ROS package that uses GST to get access to the webcam.
You configure it with your gst command. I forgot the name of the package
but a quick Google surely will make you find it.

I think that may work better for you!

nickkraus, 2020-01-18:
Not a problem with the repo as far as I can tell. Similar issues when using gscam repo.
Closing

closed 2020-01-18 by nickkraus

DanKamilov, 2020-01-24:
> Not a problem with the repo as far as I can tell. Similar issues when using gscam repo.
> Closing

how do you resolve it? I have the same problem

nickkraus, 2020-01-25:
Didn't resolve it. Havent gotten around to it in a while. Im suspicious of the ribbon cable connection although I doubt thats it considering I can get it to stream. I may try an ethernet or usb camera just to be sure.

tjcheem, 2020-02-28:
I am having the same issue. it keeps printing "select timeout" when trying to use my usb webcam. Streaming in another app works fine. Any suggestions. I did not alter any files before running.