ros-drivers/video_stream_opencv #61: Loop videofile doesn't seem to work

martiege, 2020-04-04:
Hi!

When I'm trying to use loop_videofile to loop the video I'm trying to publish, the end_frame parameter doesn't seem like it's properly changing after the subscribe method. When I check inside that method, the end_frame is correctly set from -1 to 2259, but when I try to test this within the do_capture method, the end_frame value is not changed to 2259. 

It like when the private variable config is assigned to latest_config inside the subscribe method, it is just copied, and all change is therefore dropped. 

I imagine that this wasn't intended.

awesomebytes, 2020-04-26:
Sorry, I don't fully understand your reported issue.

Are you trying to change the `stop_frame` dynamic reconfigure parameter? I've checked it right now with the small.mp4 test video, and I'm using the test launchfile in `test/test_video_file.test`. And it's working correctly for me. I'm using `rosrun rqt_reconfigure rqt_reconfigure` to change the parameter.

Are you changing the `stop_frame` parameter from the launchfile and it's not working? That is also working for me.

I can see in the video topic (I'm using `rosrun rqt_image_view rqt_image_view`) how the video loops in between the frames specified in `start_frame` and `stop_frame`.

@martiege Could you provide further information? As in, what launchfile are you using, what video file are you using (if you can share it), or even better, try to reproduce the problem with the `small.mp4` video file. How are you trying to change the `stop_frame` parameter... and anything else you think will help us understand what is going on :)

martiege, 2020-04-27:
Yes, so I've reproduced this error using the small.mp4 test video, with video_file.launch (no changes). The problem happens when the video reaches the final frame, rather than looping the video, a repeating error is produced:
`
...
[ERROR] [1587973375.310817636]: Could not capture frame
[ERROR] [1587973375.344563663]: Could not capture frame
[ERROR] [1587973375.376753030]: Could not capture frame
...
`
This continues until I stop the launch. I have debugged the code, and the problem is due to the `stop_frame` being -1. In the `subscribe` function, the `stop_frame` is set to `cv::CAP_PROP_FRAME_COUNT`. This holds through the scope of the `subscribe` function, but is reverted after, so that when the final frame is reached, it is compared against -1 rather than `cv::CAP_PROP_FRAME_COUNT`, and still thinks there are more frames. Therefore the error is produced. 

I fixed this locally by changing line 225 in `video_stream.cpp` from `VideoStreamConfig latest_config = config;` to `VideoStreamConfig& latest_config = config;`. So the basis of the problem seems to be that the local config variable is changed rather than the global used in the rest of the code. I haven't tested this on any other machine than my own, so I'm unsure whether this is a problem on my end or not. 

Another point, this is not a problem if I set the `stop_frame` to 166 in the launch file (the value the `stop_frame` is supposed to take if it is -1). If I set the `stop_frame` to 167, I will get one error per loop (as it cannot read frame 167 because it does not exist), if it is 168, I get two per loop, etc.

furushchev, 2020-04-27:
@martiege Hi! Thank you very much for opening the issue and the deep investigation.
Please could you open a pull request that includes your fix so I can check the issue?
Thank you!

awesomebytes, 2021-04-10:
This was fixed with #67

closed 2021-04-10 by awesomebytes