ros-drivers/video_stream_opencv #70: SIGSEGV (Segfault) in image_view

aytimothy, 2020-05-03:
I've had to upgrade ROS from `kinetic` to `melodic` due to the repositories being shut down. So, previously, on `kinetic`, I've had no problem running `video_stream_opencv` straight out of the box without any problems, at least for the webcam. However, come to `melodic`, it seems that `image_view` always triggers a `SIGSEGV` (segfault) and dies.

```
aytimothy@ubuntu:~$ roslaunch video_stream_opencv webcam.launch
... logging to /home/aytimothy/.ros/log/3c846264-8cdc-11ea-9c80-dca63244e983/roslaunch-ubuntu-28435.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:38679/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /webcam/webcam_stream/buffer_queue_size: 1
 * /webcam/webcam_stream/camera_info_url: 
 * /webcam/webcam_stream/camera_name: webcam
 * /webcam/webcam_stream/flip_horizontal: False
 * /webcam/webcam_stream/flip_vertical: False
 * /webcam/webcam_stream/fps: 30.0
 * /webcam/webcam_stream/frame_id: webcam_optical_frame
 * /webcam/webcam_stream/height: 0
 * /webcam/webcam_stream/loop_videofile: False
 * /webcam/webcam_stream/set_camera_fps: 30.0
 * /webcam/webcam_stream/video_stream_provider: 0
 * /webcam/webcam_stream/width: 0

NODES
  /webcam/
    webcam_image_view (image_view/image_view)
    webcam_stream (video_stream_opencv/video_stream)

ROS_MASTER_URI=http://localhost:11311

process[webcam/webcam_stream-1]: started with pid [28451]
process[webcam/webcam_image_view-2]: started with pid [28452]
[ INFO] [1588472410.274417096]: Resource video_stream_provider: 0
[ INFO] [1588472410.278582490]: Getting video from provider: /dev/video0
[ INFO] [1588472411.128208668]: Video stream provider type detected: videodevice
[ INFO] [1588472411.131100702]: Camera name: webcam
[ INFO] [1588472411.134295054]: Setting camera FPS to: 30
[ INFO] [1588472411.203305255]: Camera reports FPS: 25
[ INFO] [1588472411.207589551]: Setting buffer size for capturing frames to: 1
[ INFO] [1588472411.211263970]: Throttling to fps: 30
[ INFO] [1588472411.215136713]: Publishing with frame_id: webcam_optical_frame
[ INFO] [1588472411.220204134]: Provided camera_info_url: ''
[ INFO] [1588472411.227309579]: Flip horizontal image is: false
[ INFO] [1588472411.233874274]: Flip vertical image is: false
[ INFO] [1588472411.246008345]: using default calibration URL
[ INFO] [1588472411.246228408]: camera calibration URL: file:///home/aytimothy/.ros/camera_info/webcam.yaml
[ INFO] [1588472411.246496505]: Unable to open camera calibration file [/home/aytimothy/.ros/camera_info/webcam.yaml]
[ WARN] [1588472411.246620369]: Camera calibration file /home/aytimothy/.ros/camera_info/webcam.yaml not found.
[ INFO] [1588472411.246728011]: Opened the stream, starting to publish.
[ WARN] [1588472412.115673857]: No calibration file given, publishing a reasonable default camera info.
[ INFO] [1588472412.115804609]: The image width is: 640
[ INFO] [1588472412.115878291]: The image height is: 480
[webcam/webcam_image_view-2] process has died [pid 28452, exit code -11, cmd /opt/ros/melodic/lib/image_view/image_view image:=image_raw __name:=webcam_image_view __log:=/home/aytimothy/.ros/log/3c846264-8cdc-11ea-9c80-dca63244e983/webcam-webcam_image_view-2.log].
log file: /home/aytimothy/.ros/log/3c846264-8cdc-11ea-9c80-dca63244e983/webcam-webcam_image_view-2*.log```

I'm not sure what is going on, because I have had my `/dev/video0` available for user read/write and `cheese` seems to be able to open it fine without having the need to be `sudo`'d.

This is the stacktrace that was outputted by the crash detector:

```
Stacktrace:
 #0  0x0000ffffb4040978 in cvConvertImage () at /usr/lib/aarch64-linux-gnu/libopencv_imgcodecs.so.3.2
 #1  0x0000ffffb4017aa4 in  () at /usr/lib/aarch64-linux-gnu/libopencv_highgui.so.3.2
 #2  0x0000ffffb4019d04 in cvShowImage () at /usr/lib/aarch64-linux-gnu/libopencv_highgui.so.3.2
 #3  0x0000ffffb40168dc in cv::imshow(cv::String const&, cv::_InputArray const&) () at /usr/lib/aarch64-linux-gnu/libopencv_highgui.so.3.2
 #4  0x0000ffffb42d7b58 in image_view::ImageNodelet::windowThread() () at /opt/ros/melodic/lib//libimage_view.so
 #5  0x0000ffffb86cbfe8 in  () at /usr/lib/aarch64-linux-gnu/libboost_thread.so.1.65.1
 #6  0x0000ffffb8695088 in start_thread (arg=0xffffe915e65f) at pthread_create.c:463```

awesomebytes, 2020-05-03:
Hello,

> I've had to upgrade ROS from `kinetic` to `melodic` due to the repositories being shut down.
Kinetic's end of life is set [to April 2021](http://wiki.ros.org/Distributions). We should have one year more of it?

> So, previously, on `kinetic`, I've had no problem running `video_stream_opencv` straight out of the box without any problems, at least for the webcam. However, come to `melodic`, it seems that `image_view` always triggers a `SIGSEGV` (segfault) and dies.

We use `image_view` as a tool to visualize, but we aren't maintaining it. I can see a couple of issues in the `image_pipeline` repository describing your same error:
* https://github.com/ros-perception/image_pipeline/issues/516
* https://github.com/ros-perception/image_pipeline/issues/498

My recommendation is to record a rosbag of the topics provided by `video_stream_opencv` of short duration and check if `image_view` crashes with playing the rosbag instead of the node running. (I think it should... as `image_view` is just trying to show images from the ROS Image message format).

If you manage to reproduce it, reply to one of those issues (probably [498](https://github.com/ros-perception/image_pipeline/issues/498)) attaching your rosbag and explaining it reproduces the problem. I'm sure the developers will be very happy to have a reproducible case.

aytimothy, 2020-05-03:
> > I've had to upgrade ROS from kinetic to melodic due to the repositories being shut down.
>
> Kinetic's end of life is set to April 2021. We should have one year more of it?

That's what I thought too. Tried it on numerous new installations, all failed to find any packages that were for `kinetic`.

Going to attempt to reproduce it now. Closing as wrong place to ask, and thanks for the pointers.

closed 2020-05-03 by aytimothy

awesomebytes, 2020-05-03:
@aytimothy I just tried to install Kinetic (desktop_full, following the [official instructions](http://wiki.ros.org/kinetic/Installation/Ubuntu)) on a Docker environment with Ubuntu 16.04 and it worked no problem. I'm actually working on a Ubuntu 16.04 system and I updated some ROS Kinetic packages just last week.

I also maintain a continuous integration build of both ROS Kinetic and Melodic over Gentoo (Prefix) and they are both building (with some hickups sometimes, but they build).

Maybe you want to copy-paste the errors you are having and I may be able to give you a hand? That said, starting a project in Melodic would be probably the best approach these days :)

If your problem is related to the compressed version of the images, you may be having problems with `image_transport`. It is contained in the repo: https://github.com/ros-perception/image_common which I haven't seen any particularly interesting issue related to what you comment.

aytimothy, 2020-05-03:
Oh, I just realized, I'm on 18.04, so that would be why Kinetic wasn't working (it's for 16.x). Oops, but unrelated to the actual issue at hand.

awesomebytes, 2020-05-03:
@aytimothy makes sense then!