ros-drivers/video_stream_opencv #71: Immediate crash when using video_file.launch

matthaeusheer, 2020-05-07:
Streaming from camera.launch works fine. But: running video_file.launch does crash immediately. I tried the small example mp4 movie, the same happens with my own movies. The log given down below is actually empty.

Cheers

```
matt@xps:~$ roslaunch video_stream_opencv video_file.launch 
... logging to /home/matt/.ros/log/9e67bfc0-90ac-11ea-a57d-d83bbfa23889/roslaunch-xps-10055.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xps:33467/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /videofile/videofile_stream/buffer_queue_size: 1000
 * /videofile/videofile_stream/camera_info_url: 
 * /videofile/videofile_stream/camera_name: videofile
 * /videofile/videofile_stream/flip_horizontal: False
 * /videofile/videofile_stream/flip_vertical: False
 * /videofile/videofile_stream/fps: 30.0
 * /videofile/videofile_stream/frame_id: videofile_frame
 * /videofile/videofile_stream/height: 0
 * /videofile/videofile_stream/loop_videofile: False
 * /videofile/videofile_stream/set_camera_fps: 30.0
 * /videofile/videofile_stream/start_frame: 0
 * /videofile/videofile_stream/stop_frame: -1
 * /videofile/videofile_stream/video_stream_provider: /tmp/small.mp4
 * /videofile/videofile_stream/width: 0

NODES
  /videofile/
    videofile_image_view (image_view/image_view)
    videofile_stream (video_stream_opencv/video_stream)

auto-starting new master
process[master]: started with pid [10066]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9e67bfc0-90ac-11ea-a57d-d83bbfa23889
process[rosout-1]: started with pid [10077]
started core service [/rosout]
process[videofile/videofile_stream-2]: started with pid [10080]
process[videofile/videofile_image_view-3]: started with pid [10085]
[ INFO] [1588888166.402607238]: Initializing nodelet with 8 worker threads.
[ INFO] [1588888166.490555355]: Camera name: videofile
[ INFO] [1588888166.490587488]: Provided camera_info_url: ''
[ INFO] [1588888166.490600057]: Publishing with frame_id: videofile_frame
[ INFO] [1588888166.490617520]: Setting camera FPS to: 30
[ INFO] [1588888166.490636030]: Throttling to fps: 30
[ INFO] [1588888166.490648199]: Setting buffer size for capturing frames to: 1000
[ INFO] [1588888166.490658106]: Flip horizontal image is: false
[ INFO] [1588888166.490670467]: Flip vertical image is: false
[ INFO] [1588888166.490681324]: Video start frame is: 0
[ INFO] [1588888166.490698345]: Video stop frame is: -1
[ INFO] [1588888166.970952236]: using default calibration URL
[ INFO] [1588888166.971084083]: camera calibration URL: file:///home/matt/.ros/camera_info/videofile.yaml
[ INFO] [1588888166.971302834]: Unable to open camera calibration file [/home/matt/.ros/camera_info/videofile.yaml]
[ WARN] [1588888166.971404132]: Camera calibration file /home/matt/.ros/camera_info/videofile.yaml not found.
[ INFO] [1588888166.971790315]: Opening VideoCapture with provider: /tmp/small.mp4
[ INFO] [1588888167.016638356]: Video stream provider type detected: videofile
[ INFO] [1588888167.016784028]: Camera reports FPS: 30
[ WARN] [1588888167.053391812]: No calibration file given, publishing a reasonable default camera info.
[ INFO] [1588888167.053534105]: The image width is: 560
[ INFO] [1588888167.053590768]: The image height is: 320
[videofile/videofile_image_view-3] process has died [pid 10085, exit code -11, cmd /opt/ros/melodic/lib/image_view/image_view image:=image_raw __name:=videofile_image_view __log:=/home/matt/.ros/log/9e67bfc0-90ac-11ea-a57d-d83bbfa23889/videofile-videofile_image_view-3.log].
log file: /home/matt/.ros/log/9e67bfc0-90ac-11ea-a57d-d83bbfa23889/videofile-videofile_image_view-3*.log
```

furushchev, 2020-05-08:
@matthaeusheer Hi, it looks the program which crashes is an image viewer, not a node of this repo. Please check if `image_view` can be used standalone?

awesomebytes, 2020-05-08:
Hello @matthaeusheer seems like this is the same that is being reported here: https://github.com/ros-perception/image_pipeline/issues/516 
It's a problem of `image_view` not of `video_stream_opencv` (or so it looks like, no conclusion yet).

Hereby I'll ask you to report in that issue instead. Before doing so you can try the following things:
1. Run the node without `image_view` (in the launchfile set `visualize` to `false`) and try to check the image topic with another tool like `rqt_image_view` or Rviz.
2. If it works, then we are pretty much sure it's an issue within `image_view` (which we already suspect from the other reports https://github.com/ros-drivers/video_stream_opencv/issues/70 https://github.com/ros-drivers/video_stream_opencv/issues/64).
3. You could try to run in isolation `image_view` and see if it crashes. If it doesn't, it may have to do with some kind of initialization and race condition between the `video_stream_opencv` node publishing the images and camera info... if it does crash well... see next point.
4. As you are able to reproduce the issue. Would you be so kind to try to record a rosbag of the image topic and camera_info topic, then try to run `image_view` in isolation by playing that rosbag to check if it crashes. I've asked another user that reports the same issue in the repository of `image_view`: https://github.com/ros-perception/image_pipeline/issues/516 This is to make sure the issue has nothing to do with the format of the images published by `video_stream_opencv`. If it would actually crash `image_view`, it would be a perfect way to reproduce the bug for the maintainers to fix. I'd ask you to attach the rosbag to the [GitHub issue](https://github.com/ros-perception/image_pipeline/issues/516) I just mentioned.

Thanks for reporting this bug :)

matthaeusheer, 2020-05-08:
Hey  
I can confirm that `<arg name="visualize" value="false" />` does solve the issue, so your assumption might be right.
I'll record the bag and post it to the image_view issue.

If I might ask, I have some other questions which are not expected behavior
1) When the last frame of a video file is reached I get errors (regardless whether looping is set to true or false)
 ```[ERROR] [1588929063.892296257]: Could not capture frame
[ERROR] [1588929063.926041994]: Could not capture frame
[ERROR] [1588929063.959889255]: Could not capture frame
[ERROR] [1588929063.993355098]: Could not capture frame
...
```
2) Providing the argument for video_stream_provider, e.g. `roslaunch video_stream_opencv video_file.launch video_stream_provider:=/path/to/file.mp4` seems to have no effect. To change the video I have to go into the actual launch file and set it there, otherwise the one specified in the launchfile is used.

3)

closed 2021-04-10 by awesomebytes