ros-drivers/video_stream_opencv #72: 'Could not capture frame' error when 'stop_frame' is -1

hoonkai, 2020-05-18:
Hi

When `stop_frame` is set to `-1`, capturing the stream leads to the following error:

```
[ERROR] [1589731616.096380328]: Could not capture frame
[ERROR] [1589731616.129171038]: Could not capture frame
[ERROR] [1589731616.162531012]: Could not capture frame
[ERROR] [1589731616.195890605]: Could not capture frame
[ERROR] [1589731616.229178206]: Could not capture frame
```

as I briefly reported here: https://github.com/ros-perception/image_pipeline/issues/516#issuecomment-629823085_

Here's the result of a quick investigation. This line handles the case when `stop_frame == -1`: https://github.com/ros-drivers/video_stream_opencv/blob/master/src/video_stream.cpp#L244

where `latest_config` is a copy of `config`. But https://github.com/ros-drivers/video_stream_opencv/blob/master/src/video_stream.cpp#L140 expects a meaningful value of `stop_frame`.

I've managed to avoid the error by making `latest_config` a `VideoStreamConfig*`, but I'm not sure if that'll produce unintended effects.

furushchev, 2020-11-13:
@hoonkai Sorry for delay. Could you please check if https://github.com/ros-drivers/video_stream_opencv/issues/72 fixes the issue?

lchop, 2020-12-04:
Hello I have the same issue, when could not capture frame error is raise, the node is not able to publish again the image. I have to kill the node and re launch it.

awesomebytes, 2021-04-10:
I believe the parameter `reopen_on_read_failure` should allow to fix this, but I'm not sure if this is the same issue. I'll close for now, reopen if you think it's still an issue.

closed 2021-04-10 by awesomebytes