ros-drivers/video_stream_opencv #75: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8)

mrgransky, 2020-08-19:
running this `file.launch` in 1st terminal:

```
<launch>
	<arg name="camera_live" default="true" />
	<arg name="open_rviz" 	default="true" />
	<arg name="my_camera" 	default="Huawei_Mate_10_pro" />
	
	<node if="$(arg open_rviz)" pkg="rviz" type="rviz" name="ouster_rviz" output="screen" args="-d $(find ouster_ros)/only_huawei.rviz"/>

	<group if="$(arg camera_live)">
		<include file="$(find video_stream_opencv)/launch/camera.launch">
			<arg name="camera_name" 			value="$(arg my_camera)" />
			
			<!--<arg name="video_stream_provider" 	value="0" />
			<arg name="set_camera_fps" 			value="30"/>-->
			
			<!-- streaming url
	  		<arg name="set_camera_fps"			value="30"/> -->
			<arg name="video_stream_provider" 	value="http://192.168.43.26:8080/video?x.mjpeg" /> 

			<arg name="fps" 					value="10" />
			<arg name="frame_id" 			value="webcam_optical_frame" />
			<!-- camera.yaml-->
			<arg name="camera_info_url" 	value="file:///$(find ouster_ros)/huawei_calib.yaml" />
			<arg name="flip_horizontal" 	value="false" />
			<arg name="flip_vertical" 		value="false" />
			<arg name="buffer_queue_size" value="1000" />
			<arg name="width" 				value="0"/>
			<arg name="height" 				value="0"/>
			<arg name="visualize" 			value="false" />
		</include>
	</group>
</launch>
```
and `rosbag record -a --duration=20s -O bag_file` in 2nd terminal, I get the following error streaming on my 1st terminal:

`[ERROR] [1597829296.595991253]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8)`

Camera Setup:
Android phone, Huawei Mate10 Pro 
Android App: IP webcam (tried video resolution: `640x480`,` 3264x1632`)

awesomebytes, 2020-08-19:
I saw you already posted there but... This is not about video_stream_provider but about image_transport, as reported in this link:
https://answers.ros.org/question/251403/image_transport-rosbag-issue/

mrgransky, 2020-08-19:
so is there anyway to fix it though?

awesomebytes, 2020-08-19:
@mrgransky Well, it's not really broken, or does it crash rosbag?
I've never used `rosbag record -a` as usually I only want a subset of topics (for example, that will record `/rosout` which is uncommon to want it recorded). My recommendation would be to instead do `rosbag record /Huawei_Mate_10_pro/image_raw /Huawei_Mate_10_pro/camera_info` and any other topic of your interest.

 Otherwise, the answer in that link gives you a solution, add to the roslaunch:
```xml
        <!-- disable compressed depth plugin for image transport -->
        <group ns="image_raw">
            <rosparam param="disable_pub_plugins">
              - 'image_transport/compressedDepth'
            </rosparam>
        </group>
```

You could copy https://github.com/ros-drivers/video_stream_opencv/blob/master/launch/camera.launch and make it look like:
```xml
<?xml version="1.0"?>
<launch>
  	<arg name="camera_name" default="camera" />
  	<!-- video_stream_provider can be a number as a video device or a url of a video stream -->
  	<arg name="video_stream_provider" default="0" />
  	<!-- set camera fps to -->
  	<arg name="set_camera_fps" default="30" />
  	<!-- set buffer queue size of frame capturing to -->
  	<arg name="buffer_queue_size" default="100" />
  	<!-- frames per second to query the camera for -->
  	<arg name="fps" default="30" />
  	<!-- frame_id for the camera -->
  	<arg name="frame_id" default="$(arg camera_name)" />
  	<!-- By default, calibrations are stored to file://${ROS_HOME}/camera_info/${NAME}.yaml
  	To use your own fill this arg with the corresponding url, e.g.:
  	"file:///$(find your_camera_package)/config/your_camera.yaml" -->
    <arg name="camera_info_url" default="" />
  	<!-- flip the image horizontally (mirror it) -->
  	<arg name="flip_horizontal" default="false" />
  	<!-- flip the image vertically -->
  	<arg name="flip_vertical" default="false" />
    <!-- force width and height, 0 means no forcing -->
    <arg name="width" default="0"/>
    <arg name="height" default="0"/>
    <!-- enable looping playback, only if video_stream_provider is a video file -->
    <arg name="loop_videofile" default="false" />
    <arg name="start_frame" default="0"/>
    <arg name="stop_frame" default="-1"/>
    <!-- if show a image_view window subscribed to the generated stream -->
	<arg name="visualize" default="false"/>

   
   	<!-- images will be published at /camera_name/image with the image transports plugins (e.g.: compressed) installed -->
   	<group ns="$(arg camera_name)">
	    <node pkg="video_stream_opencv" type="video_stream" name="$(arg camera_name)_stream" output="screen"> 
	    	<remap from="camera" to="image_raw" />
	    	<param name="camera_name" type="string" value="$(arg camera_name)" />
	        <param name="video_stream_provider" type="string" value="$(arg video_stream_provider)" />
	        <param name="set_camera_fps" type="double" value="$(arg set_camera_fps)" />
	        <param name="buffer_queue_size" type="int" value="$(arg buffer_queue_size)" />
	        <param name="fps" type="double" value="$(arg fps)" />
	        <param name="frame_id" type="string" value="$(arg frame_id)" />
	        <param name="camera_info_url" type="string" value="$(arg camera_info_url)" />
	        <param name="flip_horizontal" type="bool" value="$(arg flip_horizontal)" />
	        <param name="flip_vertical" type="bool" value="$(arg flip_vertical)" />
                <param name="loop_videofile" type="bool" value="$(arg loop_videofile)" />
                <param name="start_frame" type="int" value="$(arg start_frame)" />
                <param name="stop_frame" type="int" value="$(arg stop_frame)" />
	        <param name="width" type="int" value="$(arg width)" />
	        <param name="height" type="int" value="$(arg height)" />
	    </node>
            <!-- disable compressed depth plugin for image transport -->
            <group ns="image_raw">
                <rosparam param="disable_pub_plugins">
                  - 'image_transport/compressedDepth'
                </rosparam>
            </group>

	    <node if="$(arg visualize)" name="$(arg camera_name)_image_view" pkg="image_view" type="image_view">
	    	<remap from="image" to="image_raw" />
	    </node>
	</group>

</launch>
```
To apply that solution. Note that I haven't tested this, but it should work.

Good luck!

closed 2021-04-10 by awesomebytes