ros-drivers/video_stream_opencv #78: thread error if set "reopen_on_read_failure" to be true

CanboYe, 2020-10-27:
Because I got Error "Could not capture frame" sometimes, I set "reopen_on_read_failure" to be true.

However, this will result in the thread error as follows and shut down the camera node.

```
[ERROR] [1603808756.444539510]: Could not capture frame
[ WARN] [1603808756.444615546]: trying to reopen the device
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::thread_resource_error> >'
  what():  boost thread: trying joining itself: Resource deadlock avoided
```

Btw, I will always get frame dropping and I cannot find the reasons. Is that common? Three kinds of dropping messages are as follows.
```
[rtsp @ 0x7fbfc802db60] RTP timestamps don't match.
[rtsp @ 0x7fbfc802db60] Received packet without a start chunk; dropping frame.
[rtsp @ 0x7fbfc802db60] Missing packets; dropping frame.
```

Thanks a lot!

awesomebytes, 2021-04-10:
RTP streams do lose data in my experience. I am not happy that video_stream_opencv doesn't deal with that problem gracefully enough, but I don't have a test environment to reproduce the problem.

Did you find a fix?

rolker, 2021-05-21:
I'm encountering the same problem and have setup a way to reproduce the failure in a debugging session. Here's what's happening:

In video_stream.cpp's  do_capture() method, we get to line 129 where unsubscribe() is called.
 At line 318 in the unsubscribe() method, capture_thread.join() is called which causes the exception.

It's my first time looking at this code base, so I don't a solution yet, but it seems like we either need to not restart the thread when resubscribing or exit the thread before doing the unsubscribe/resubscribe.

I'll report back if I come up with a solution, but if I don't, I hope the above is useful to someone.