ros-drivers/video_stream_opencv #79: terminate called after throwing an instance of 'std::runtime_error'

Adidushi, 2020-11-08:
I am trying to stream an ip camera over rtsp using video_stream_opencv. 
I have been able to successfully visualize the camera using VLC with the command:

`vlc --network-caching=500 rtsp://admin:admin@10.201.41.101:5541`

However any attempts to stream the camera with video_stream has failed. Here is the result of attempting to run video_stream_opencv:

```
roslaunch video_stream_opencv camera.launch video_stream_provider:="rtsp://admin:admin@10.201.41.101:5541" visualize:=true
... logging to /home/user/.ros/log/824e7b1e-f3a1-11ea-8e38-509a4c4c9a82/roslaunch-user-OptiPlex-7050-7261.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/user/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://user-OptiPlex-7050:44619/

SUMMARY
========

PARAMETERS
 * /camera/camera_stream/buffer_queue_size: 100
 * /camera/camera_stream/camera_info_url: 
 * /camera/camera_stream/camera_name: camera
 * /camera/camera_stream/flip_horizontal: False
 * /camera/camera_stream/flip_vertical: False
 * /camera/camera_stream/fps: 30.0
 * /camera/camera_stream/frame_id: camera
 * /camera/camera_stream/height: 0
 * /camera/camera_stream/loop_videofile: False
 * /camera/camera_stream/set_camera_fps: 30.0
 * /camera/camera_stream/video_stream_provider: rtsp://admin:admi...
 * /camera/camera_stream/width: 0
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /camera/
    camera_image_view (image_view/image_view)
    camera_stream (video_stream_opencv/video_stream)

auto-starting new master
process[master]: started with pid [7271]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 824e7b1e-f3a1-11ea-8e38-509a4c4c9a82
process[rosout-1]: started with pid [7282]
started core service [/rosout]
process[camera/camera_stream-2]: started with pid [7289]
process[camera/camera_image_view-3]: started with pid [7290]
[ INFO] [1599768559.808352235]: Resource video_stream_provider: rtsp://admin:admin@10.201.41.101:5541
[ INFO] [1599768559.808787364]: Getting video from provider: rtsp://admin:admin@10.201.41.101:5541
[ INFO] [1599768589.924488669]: Video stream provider type detected: rtsp_stream
[ INFO] [1599768589.926685254]: Camera name: camera
[ INFO] [1599768589.928740132]: Setting camera FPS to: 30
[ INFO] [1599768589.928824345]: Camera reports FPS: 90000
[ INFO] [1599768589.930779402]: Setting buffer size for capturing frames to: 100
[ INFO] [1599768589.932821669]: Throttling to fps: 30
[ INFO] [1599768589.934871704]: Publishing with frame_id: camera
[ INFO] [1599768589.936872284]: Provided camera_info_url: ''
[ INFO] [1599768589.938861274]: Flip horizontal image is: false
[ INFO] [1599768589.940802556]: Flip vertical image is: false
[ INFO] [1599768589.948160366]: using default calibration URL
[ INFO] [1599768589.948329780]: camera calibration URL: file:///home/user/.ros/camera_info/camera.yaml
[ INFO] [1599768589.948532653]: Unable to open camera calibration file [/home/user/.ros/camera_info/camera.yaml]
[ WARN] [1599768589.948644310]: Camera calibration file /home/user/.ros/camera_info/camera.yaml not found.
[ INFO] [1599768589.948729846]: Opened the stream, starting to publish.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not capture frame!
[camera/camera_stream-2] process has died [pid 7289, exit code -6, cmd /opt/ros/melodic/lib/video_stream_opencv/video_stream camera:=image_raw __name:=camera_stream __log:=/home/user/.ros/log/824e7b1e-f3a1-11ea-8e38-509a4c4c9a82/camera-camera_stream-2.log].
log file: /home/user/.ros/log/824e7b1e-f3a1-11ea-8e38-509a4c4c9a82/camera-camera_stream-2*.log
```

Changing the FPS and buffer size seem to change the frequency of it but I'm unsure as to what is happening.

awesomebytes, 2020-11-12:
Hello @Adidushi, the error `what():  Could not capture frame!` makes me think that it's just the OpenCV backend not able to grab an image.

I invite you to try the test script https://github.com/ros-drivers/video_stream_opencv/blob/master/scripts/test_video_resource.py with your rtsp URL and see if you find the same behavior. As you may see, the script is minimal, if it crashes in the same fashion, it means the OpenCV backend is not able to handle that stream (at least without setting more options to the stream? but I am not familiar with what options a rtsp stream provider may have).

If this doesn't work, I would try to use gstreamer to decode your stream. I found this link that has an example of a client commandline (under RTSP clients): https://developer.ridgerun.com/wiki/index.php?title=GstRtspSink_-_Live_File_Streaming
Another example here: https://github.com/google-coral/examples-camera/issues/16

If that works, you may want to look into using gscam http://wiki.ros.org/gscam with your custom gstreamer configuration string

Good luck! (And please, reply here with your findings so the community benefits from them if you have the time).

Adidushi, 2020-11-12:
Hi, the issue was that my RTSP stream wasn't very stable. I ended up writing a custom script that just restarts the node whenever it crashes.

Thanks!

closed 2020-11-12 by Adidushi