ros-drivers/video_stream_opencv #84: noetic?

PfeifferMicha, 2020-12-07:
Hi,

as far as I can tell, the newest officially supported version is melodic. Is there a reason for this? In my build, the newest version of this works fine if I clone it into my (noetic) catkin workspace.
Are there plans to officially support noetic as well?
Otherwise, is there an official replacement?

Thanks

awesomebytes, 2020-12-07:
Pull request to release on noetic done: https://github.com/ros/rosdistro/pull/27638

I don't know when is the next sync, but it will be released sometime soon :)

Thanks for the heads up! (I also re-released on melodic to get the latest fixes there).

PfeifferMicha, 2020-12-07:
Great, thanks a lot for the quick reaction!

awesomebytes, 2020-12-07:
@PfeifferMicha Unfortunately, the buildfarm failed to build it for Noetic:
http://build.ros.org/job/Nbin_uF64__video_stream_opencv__ubuntu_focal_amd64__binary/2/console

The error itself is:
```
23:12:42 [ 50%] Building CXX object CMakeFiles/video_stream.dir/src/video_stream.cpp.o
23:12:42 /usr/lib/ccache/c++  -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"video_stream_opencv\" -I/opt/ros/noetic/include -I/opt/ros/noetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -isystem /usr/include/opencv4  -g -O2 -fdebug-prefix-map=/tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5=. -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2   -std=c++11 -o CMakeFiles/video_stream.dir/src/video_stream.cpp.o -c /tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5/src/video_stream.cpp
23:12:49 /tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5/src/video_stream.cpp: In function ‘void do_capture(ros::NodeHandle&)’:
23:12:49 /tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5/src/video_stream.cpp:107:38: error: ‘CV_CAP_PROP_FRAME_COUNT’ was not declared in this scope
23:12:49   107 |             frame_counter == cap.get(CV_CAP_PROP_FRAME_COUNT))
23:12:49       |                                      ^~~~~~~~~~~~~~~~~~~~~~~
23:12:49 /tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5/src/video_stream.cpp: In function ‘int main(int, char**)’:
23:12:49 /tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5/src/video_stream.cpp:190:13: error: ‘CV_CAP_PROP_FPS’ was not declared in this scope
23:12:49   190 |     cap.set(CV_CAP_PROP_FPS, set_camera_fps);
23:12:49       |             ^~~~~~~~~~~~~~~
23:12:49 /tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5/src/video_stream.cpp:260:17: error: ‘CV_CAP_PROP_FRAME_WIDTH’ was not declared in this scope
23:12:49   260 |         cap.set(CV_CAP_PROP_FRAME_WIDTH, width_target);
23:12:49       |                 ^~~~~~~~~~~~~~~~~~~~~~~
23:12:49 /tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5/src/video_stream.cpp:261:17: error: ‘CV_CAP_PROP_FRAME_HEIGHT’ was not declared in this scope
23:12:49   261 |         cap.set(CV_CAP_PROP_FRAME_HEIGHT, height_target);
23:12:49       |                 ^~~~~~~~~~~~~~~~~~~~~~~~
23:12:49 make[4]: *** [CMakeFiles/video_stream.dir/build.make:66: CMakeFiles/video_stream.dir/src/video_stream.cpp.o] Error 1
23:12:49 make[4]: Leaving directory '/tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5/obj-x86_64-linux-gnu'
23:12:49 make[3]: *** [CMakeFiles/Makefile2:168: CMakeFiles/video_stream.dir/all] Error 2
23:12:49 make[3]: Leaving directory '/tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5/obj-x86_64-linux-gnu'
23:12:49 make[2]: *** [Makefile:133: all] Error 2
23:12:49 make[2]: Leaving directory '/tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5/obj-x86_64-linux-gnu'
23:12:49 dh_auto_build: error: cd obj-x86_64-linux-gnu && make -j1 returned exit code 2
23:12:49 make[1]: *** [debian/rules:38: override_dh_auto_build] Error 25
23:12:49 make[1]: Leaving directory '/tmp/binarydeb/ros-noetic-video-stream-opencv-1.1.5'
23:12:49 make: *** [debian/rules:22: build] Error 2
```

I thought it would just go through based on your comment, can you take a look?

awesomebytes, 2020-12-08:
Oh, I see what happened. It is using a very old release. I need to make a new version first. My bad.

awesomebytes, 2020-12-08:
Ok, done again: https://github.com/ros/rosdistro/pull/27648 (and melodic too https://github.com/ros/rosdistro/pull/27649).

I did ran a pre-release but it failed before actually getting to build the code for some issue with my machine. It was late so I had no time to look into it.

awesomebytes, 2020-12-08:
I'll need to get this through first: https://github.com/ros-drivers/video_stream_opencv/pull/85

PfeifferMicha, 2020-12-08:
Ah, the errors above seem to be OpenCV errors. It's possible that the machine I was running it on had both OpenCV 4 and an older version installed, maybe that's why it compiled without issues for me?
It looks like the correct version to support for noetic would be version 4.2: https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025
I think I have access to a machine with only OpenCV 4.2 on it, if you need me to I can do some tests there. And possibly do some porting, if it's not too much (I'm no OpenCV expert).

awesomebytes, 2020-12-10:
@PfeifferMicha if you have time to look into it, that would be great. Note that I started a bit of that effort in this branch:
https://github.com/ros-drivers/video_stream_opencv/tree/bugfix-fix-mjpg_server-test

Where I patched up a little bug and enabled automated testing on noetic. I believe that If you make Pull Request, the CI will trigger on each commit, so you may have a bit less pain testing if your changes work for a release (i.e. it builds and it successfully runs the tests).

Otherwise, I'd recommend you to try to fix it inside of a docker container instead of a on a native system. I can try to help you to set that up if you have difficulties.