ros-drivers/video_stream_opencv #92: The camera images turned permanently grey and black

Jendker, 2021-04-02:
Unfortunately one time after using the video_stream_opencv in docker all the cameras turned grey and black and restarting the machine does not solve it. This effect persists also when using other apps (Zoom etc.). If we plug in another camera, it works fine.

I wanted to ask what is the reason behind setting all the different device settings in the `subscribe` function. https://github.com/ros-drivers/video_stream_opencv/blob/68cde068bed55488a9d452ee12a9158838843589/src/video_stream.cpp#L293 I am considering if the failed call of these functions could have a permanent effect on the hardware? I'm not sure what otherwise could have caused it.

Below image presents how the output of the cameras looks now.

[image]

awesomebytes, 2021-04-10:
Hello @Jendker. In this package we set a reasonable set of defaults. Which now that I think about it, maybe they should not be touched up until a user changes them. I may give a go to implementing that.

Anyways, to solve your issue, I propose you use `v4l2-ctl` (`sudo apt-get install v4l-utils`) to discover the defaults of your camera, and then you set them. I must say I'm surprised your camera remembers them after losing power. It works like:
```bash
v4l2-ctl --all

Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : Integrated_Webcam_HD: Integrate
	Bus info      : usb-0000:00:14.0-7
	Driver version: 5.4.94
	Capabilities  : 0x84A00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'YUYV'
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 614400
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Limited Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=0 value=0
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=-1200
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=300 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=1 max=8 step=1 default=1 value=1
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=1
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=10 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=3 value=3
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
              exposure_absolute 0x009a0902 (int)    : min=10 max=625 step=1 default=166 value=166 flags=inactive
```

There you can see the different parameters, like `brigthness` and `contrast`.

Then you set them to the default you read there like:
```bash
v4l2-ctl --set-ctrl brightness=0
```

Jendker, 2021-04-12:
That has helped, thank you!

closed 2021-04-12 by Jendker

zoombinis, 2021-04-23:
@awesomebytes Can you clarify how to correct the config?

For instance, using `v4l2` it is showing my camera image dimensions as:

>Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1024/720

But actually I have configured the camera to display `640x480`

But `video_stream_opencv` logs the image size as `640x320`, which I think is compromising my camera calibration I'm trying to perform.

How can I force the correct image height, or does this not really make any impact?

awesomebytes, 2021-04-23:
You can set the desired camera size in the launch file settings of this
package, they will be attempted to be set via the opencv API.

Some cameras do ignore it, though

zoombinis, 2021-04-23:
@awesomebytes unfortunately the launch file setting is not working for me. I filed a new issue for this. Is there no way to correctly set the width? I haven't seen this issue before with other ros nodes consuming this camera stream.

https://github.com/ros-drivers/video_stream_opencv/issues/99

dhgarcia, 2023-02-11:
I was having this same problem so thanks @Jendker and @awesomebytes, and thanks to @machinekoder for highlighting the OpenCV change that creates the issue.

If it helps to anybody I found a solution.
After reading through https://github.com/opencv/opencv/blob/master/modules/videoio/src/cap_v4l.cpp#L379 
This behaviour can be solved by just setting the environment variable OPENCV_VIDEOIO_V4L_RANGE_NORMALIZED to true.