ros-drivers/video_stream_opencv #100: missing calibration and camera_model yaml param not read

zoombinis, 2021-04-23:
Hi,

I've used this node to publish RTSP to a ROS topic in order to calibrate the camera using ROS camera_calibrate node. Calibration generated the following yaml output below. However, upon loading this calibration file there is no noticeable change in reducing the distortion. Also, a warning is logged that it couldn't find the `distortion_model` parameter in the yaml, however that should be listed as `camera_model`, at least in ROS melodic

```yaml
image_width: 640
image_height: 480
camera_name: rugged_hd_ip
camera_matrix:
  rows: 3
  cols: 3
  data: [ 543.2134 ,    0.     ,  321.01119,
            0.     ,  543.06716,  190.99665,
            0.     ,    0.     ,    1.     ]
camera_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.480672, 0.184891, -0.001901, 0.002792, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [ 1.,  0.,  0.,
          0.,  1.,  0.,
          0.,  0.,  1.]
projection_matrix:
  rows: 3
  cols: 4
  data: [ 429.48141,    0.     ,  325.60516,    0.     ,
            0.     ,  512.12939,  191.97364,    0.     ,
            0.     ,    0.     ,    1.     ,    0.     ]
```

Without calibration:
[image]

With calibration (and even if I change `camera_model` to `distortion_model` still it makes no difference):
```shell
roslaunch video_stream_opencv camera.launch video_stream_provider:=rtsp://192.168.1.1:554/h264 camera_info_url:='file:////root/git/adas/rugged_hd_ip_cal.yaml' width:=640 height:=480 camera_name:=rugged_hd_ip
```

[image]

awesomebytes, 2021-04-24:
You need to use a rectifier node/nodelet to apply the calibration into a
rectified image. Declaring the calibration in the yaml file enables using
nodes of that kind.

zoombinis, 2021-04-24:
@awesomebytes I am using image_proc to rectify. But I needed to still correct the following yaml config:

>image_width: 640
image_height: 480
distortion_model: plumb_bob

I think it's a bug expecting "distortion_model" in ros melodic? Also the image height was set wrong (360)

awesomebytes, 2021-04-25:
@zoombinis not sure what version of camera_calibration you are using, the latest seem to correctly write `distortion_model`: https://github.com/ros-perception/image_pipeline/blob/noetic/camera_calibration/src/camera_calibration/calibrator.py#L652

Seems it was fixed in August 2020: https://github.com/ros-perception/image_pipeline/pull/580

Oh wait! I saw that the last release was Previous that merge: https://github.com/ros-perception/image_pipeline/releases/tag/melodic-1.15.0

You could use the latest version from source and you'll have that bug fixed. This is not an issue from this repo, though. I'll close this issue (I answered in your other issues too as far as I could).

closed 2021-04-25 by awesomebytes