ros-drivers/video_stream_opencv #98: Strange errors consuming RTSP h264 stream

zoombinis, 2021-04-23:
The following error logs I get while consuming h264 RTSP stream from a [visual engineering HD Rugged IP camera](https://www.visualengineering.co.uk/hd-ip-rugged-ip-camera/p/1159) with the latest firmware installed. I did not run with ros DEBUG on because the debug messages that print the frame count would truncate my terminals history if left running for long enough (these errors are intermittent).

Any thoughts on what could be causing them? Or is it likely an issue with the camera and not this repo? The camera is just on my desk - no motion or environmental factors:

>[h264 @ 0x7f6fdc031380] Invalid level prefix
[h264 @ 0x7f6fdc031380] error while decoding MB 39 22
[h264 @ 0x7f6fdc031380] top block unavailable for requested intra mode -1
[h264 @ 0x7f6fdc031380] error while decoding MB 0 0
[h264 @ 0x7f6fdc031380] Frame num change from 0 to 215
[h264 @ 0x7f6fdc031380] decode_slice_header error
Invalid UE golomb code
[h264 @ 0x7f6fdc1d4240] mb_type -1094995529 in I slice too large at 0 0
[h264 @ 0x7f6fdc1d4240] error while decoding MB 0 0
[h264 @ 0x7f6fdc1f0580] slice type 32 too large at 0
[h264 @ 0x7f6fdc1f0580] decode_slice_header error
[h264 @ 0x7f6fdc1f0580] no frame!
[h264 @ 0x7f6fdc2b4ec0] corrupted macroblock 39 22 (total_coeff=-1)
[h264 @ 0x7f6fdc2b4ec0] error while decoding MB 39 22
[h264 @ 0x7f6fdc1f0580] corrupted macroblock 39 22 (total_coeff=-1)
[h264 @ 0x7f6fdc1f0580] error while decoding MB 39 22
[h264 @ 0x7f6fdc2b4ec0] corrupted macroblock 31 16 (total_coeff=-1)
[h264 @ 0x7f6fdc2b4ec0] error while decoding MB 31 16
[NULL @ 0x7f6fdc019580] reference count 1 overflow
[h264 @ 0x7f6fdc279c20] reference count overflow
[h264 @ 0x7f6fdc279c20] decode_slice_header error
[h264 @ 0x7f6fdc279c20] no frame!
[h264 @ 0x7f6fdc279c20] out of range intra chroma pred mode
[h264 @ 0x7f6fdc279c20] error while decoding MB 38 22

awesomebytes, 2021-04-25:
Hello @zoombinis . Unfortunately this package is just a wrapper of the OpenCV API, which itself, wraps a few backends (uses whichever you have available) to offer its functionality.

If I google one of those errors I get results like:
* [Having a too-slow consuming part of the RTSP stream gives more errors](https://stackoverflow.com/questions/28276949/opencv-rtsp-stream-corrupted-macroblock) If the computer where you are running this is not able to deal with the uncompression of the h264 stream (does video_stream_opencv consume a lot of CPU in your machine?) may be that. That thread also talks about adding `?tcp` to the stream to force tcp to not have corrupted frames (but gives delay apparently).
* [This other thread talks about ffmpeg having issues with RTSP via udp](https://trac.ffmpeg.org/ticket/285), another issue in this repo reported a way to switch to UDP instead: https://github.com/ros-drivers/video_stream_opencv/issues/86 but maybe it contains enough keywords to find some workaround here

You could also try to use the ROS package gscam instead of this one. Good luck!

aarif-saifi63, 2024-11-29:
I am also having the same issues when i am running my opencv application on rtsp.

[h264 @ 0x61555d72b380] Invalid level prefix
[h264 @ 0x61555d72b380] error while decoding MB 29 20
[h264 @ 0x61555caafe40] cbp too large (3199971767) at 39 11
[h264 @ 0x61555caafe40] error while decoding MB 39 11
[h264 @ 0x61555d72b380] corrupted macroblock 15 29 (total_coeff=-1)
[h264 @ 0x61555d72b380] error while decoding MB 15 29
[h264 @ 0x61555ca70cc0] cbp too large (3199971767) at 2 19
[h264 @ 0x61555ca70cc0] error while decoding MB 2 19
[h264 @ 0x61555d312300] Invalid level prefix
[h264 @ 0x61555d312300] error while decoding MB 4 14
[h264 @ 0x61555d341440] Invalid level prefix
[h264 @ 0x61555d341440] error while decoding MB 27 28
[h264 @ 0x61555d63c280] Invalid level prefix
[h264 @ 0x61555d63c280] error while decoding MB 38 23
[h264 @ 0x61555d3405c0] corrupted macroblock 15 17 (total_coeff=-1)
[h264 @ 0x61555d3405c0] error while decoding MB 15 17
[h264 @ 0x61555d2c3080] corrupted macroblock 19 6 (total_coeff=-1)
[h264 @ 0x61555d2c3080] error while decoding MB 19 6
[h264 @ 0x61555d72b380] Invalid level prefix
[h264 @ 0x61555d72b380] error while decoding MB 36 18
[h264 @ 0x61555caafe40] Invalid level prefix
[h264 @ 0x61555caafe40] error while decoding MB 35 16
[h264 @ 0x61555ca3ce40] Invalid level prefix
[h264 @ 0x61555ca3ce40] error while decoding MB 12 14
[h264 @ 0x61555ca70cc0] corrupted macroblock 1 19 (total_coeff=-1)
[h264 @ 0x61555ca70cc0] error while decoding MB 1 19
[h264 @ 0x61555d65d200] negative number of zero coeffs at 38 16
[h264 @ 0x61555d65d200] error while decoding MB 38 16