ros-drivers/video_stream_opencv #103: Install with ROS

paulinaacostac, 2021-05-16:
Im trying to install it with ROS with the command sudo apt-get install ros-melodic-video-stream-opencv but I get error:

Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-video-stream-opencv/ros-melodic-video-stream-opencv_1.1.6-1bionic.20210416.182907_arm64.deb Could not resolve 'packages.ros.org'

Im trying to use the command rosrun video_stream_opencv test_video_resource.py 0

Running the script with Python only works perfectly fine

awesomebytes, 2021-05-16:
Either there is a problem with your DNS that can't resolve that URL, or the
url itself is down (check it in the browser) or you may need to do `sudo
apt update` and try again.

The package is hosted with the rest of the ROS packages, so we can't do
anything from here. You could use the package from source, of course.

paulinaacostac, 2021-05-17:
Thank you, it was indeed an internet issue. 
sudo apt-get install ros-melodic-video-stream-opencv
was the command I used and worked.

closed 2021-05-17 by paulinaacostac