ros-drivers/video_stream_opencv #107: Published Video Has No Sound

Faizi-AdnanFahad, 2021-06-17:
HI,

I was able to publish a video file into my robot's monitor but there's no sound. Is there any extra parameters that would do it or it doesn't support sound? if it doesn't is there any other node that publishes both the video and sound?

Thanks

awesomebytes, 2021-06-17:
There's no sound support in this package

Faizi-AdnanFahad, 2021-06-17:
Thanks for the quick replay.
Is there any other package that publishes both video and audio of the video?

awesomebytes, 2021-06-17:
I'm sorry, but, what are you trying to achieve exactly?

Sounds like you just want to play a video on a screen somehow?

Faizi-AdnanFahad, 2021-06-17:
[image]

Faizi-AdnanFahad, 2021-06-17:
[image]

So I have this robot with a monitor attached to it which I have used the gazebo video plugin to give it the ability to subscribe to images/videos published to it. But when I want to publish a video that has sound too, I want to output the sound as well as its publishing. is that a little more clear?

Thanks,

awesomebytes, 2021-06-17:
Gazebo has no sound simulation as far as I know. All you can do is play the
audio on your machine.

You could use VLC without screen output to achieve that, for example:
https://askubuntu.com/a/368751

Faizi-AdnanFahad, 2021-06-17:
Okay Thanks Sam!