ros-drivers/video_stream_opencv #109: Publishing from a gimble camera

hichamhendy, 2021-07-20:
I am trying to publish a topic from a rtsp link but I got the error 

[FATAL] [1626767105.585861559]: Invalid 'video_stream_provider': rtsp://192.168.0.201:554/live0

When I try with vlc or gstream it works fine through the following command 
gst-launch-1.0 -v rtspsrc location=rtsp://192.168.0.201:554/live0 drop-on-latency=true use-pipeline-clock=true do-retransmission=false latency=0 protocols=GST_RTSP_LOWER_TRANS_UDP ! rtph264depay ! h264parse ! avdec_h264 ! autovideosink sync=true

but anyway, i want to to publish the stream on topic: By the way I tried with webcam and it also works fine! Only with rtsp

awesomebytes, 2021-07-20:
If that gstreamer string works for you, you could try to use the ROS
package gscam

Not sure why you get that error!

hichamhendy, 2021-07-20:
Do I need to provide you with more info about the error?

awesomebytes, 2021-07-20:
Maybe show the launchfile?

You can also try to use the Python test file to see if that works. The
backend is also opencv so it should provide the same results.

The error seems to from from opencv, so maybe there's some trick to format
the rtsp stream string?

hichamhendy, 2021-07-20:
Here it is 

<?xml version="1.0"?>
<launch>
   
   <include file="$(find video_stream_opencv)/launch/camera.launch" >
   		
	  	<arg name="camera_name" value="rtsp2" />
	  	
	  	<arg name="video_stream_provider" value="rtsp://192.168.0.201:554/live0" />
	  	
	  	
      	
      	<arg name="buffer_queue_size" value="1000" />
	  	
	  	<arg name="fps" value="30" />
	  	
	  	<arg name="frame_id" value="rtsp_frame" />
	  	
	  	<arg name="camera_info_url" value="" />
	  	
	  	<arg name="flip_horizontal" value="false" />
	  	
	  	<arg name="flip_vertical" value="false" />
	  	
	  	<arg name="visualize" value="true" />
   </include>
</launch>

I didn't change much in there! only added my rtsp link, Should I have done something else?

From test_video_resource, I get:
resource.py rtsp://192.168.0.201:554/live0
('Trying to open resource: ', 'rtsp://192.168.0.201:554/live0')
Error opening resource: rtsp://192.168.0.201:554/live0
Maybe opencv VideoCapture can't open it

hichamhendy, 2021-07-20:
Here it is: arg name="video_stream_provider" value="rtsp://192.168.0.201:554/live0" 

I didn't change much in there! only added my rtsp link, Should I have done something else?

From test_video_resource, I get:
resource.py rtsp://192.168.0.201:554/live0
('Trying to open resource: ', 'rtsp://192.168.0.201:554/live0')
Error opening resource: rtsp://192.168.0.201:554/live0
Maybe opencv VideoCapture can't open it

awesomebytes, 2021-07-20:
Looks like opencv can't open it!

Too bad. I hope gscam works for you